jgarff/rpi_ws281x #523: pigpio interfering with PCM operation

sam-vdp, 2023-07-29:
Posting this mostly in the hope that it maybe helpful for others (I just spent a day diagnosing this):

pigpio by default uses the PCM hardware in a way that breaks rpi_ws281x use of the PCM channel. ws281x will kind of work, but there will be sporadic errors in the data sent to the LEDs. Not sure if ws281x could detect if anything else is using the PCM hardware and raise an error?

As a solution, pigpio can be configured to use PWM hardware instead by using gpioCfgClock

MrNNP, 2024-10-29:
I just spent a few days debugging this before I found this issue. 

My use-case needs both precise servo control, LEDs, and analog audio. Both pigpiod and ws281x are configured to use PCM, (ws281x on SPI GPIO10, servo on GPIO12) and I am running into the data error issue you describe. The LEDs will go all white instead of the configuration set by ws281x. 

Switching pigpiod to hardware PWM isn't an option because I require analog audio. 

However, while testing I noticed that the LEDs work perfectly when the PI is above a certain level of CPU usage. `apt install`  worked, but the command I settled on to stress the CPU was `yes < /dev/null &`. It has now been running for 6+ hours with zero issues, along with working servo control and analog audio. 

My use-case is satisfied, but leaving an endless loop running in the background seems like a patch for a different underlying issue. It also keeps the PI CPU hot and uses more power. 

I tried manually decreasing/increasing the process priority of ws281x with `renice` to see if I could get it to work but nothing worked other than keeping the CPU at a high % utilization.  

I am running an RPI 4B with the latest 64-bit Raspbian lite image (Kernel version: 6.6, Debian version: 12 (bookworm)). 

Everything is working with my current setup, but wasting CPU cycles doesn't seem like the optimal solution to this.

I don't have as much experience with the Linux kernel scheduler, which I assume is playing some part in this, and my admittedly minimal investigation didn't yield anything.

jgarff, 2024-10-30:
I would very much recommend not using the same PCM hardware simultaneously between 2 pieces of software that know nothing about each other.  That is going to be fraught with race conditions and hardware conflicts, which is evident by you description of running another task making things better.  This is likely because time is scheduled between users of the PCM hardware adding enough time for it to work.  I suspect that this won't work all the time though.

My recommendation is to use PCM and PWM or SPI instead.  Then there is no hardware conflict.  If you still need sound, in the PWM case, I would recommend a cheap USB sound device, or bluetooth.

It's not going to be possible for the library to detect another user of the hardware and then somehow timeslice the use of the device with it.  Hope that helps.

closed 2024-10-30 by jgarff