jgarff/rpi_ws281x #526: RPi 3 Model A+ Rev 1.1 - Hardware revision is not supported

avidesk, 2023-10-24:
Latest hardware rev of the Pi 3 Model A+ is not yet supported. Please add the following to rpihw.c

`,
    {
        .hwver  = 0x9020e1,
        .type = RPI_HWVER_TYPE_PI2,
        .periph_base = PERIPH_BASE_RPI2,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 3 Model A Plus Rev 1.1",
    }`

LaWfulN, 2023-11-08:
Hello, I manually added this to my rpihw.c file but I'm still getting the -3 Hardware revision not supported runtime error. I confirmed I am on hwver 9020e1, are there any other configurations I might be missing?

(Brand new user, be kind)

rnlgreen, 2023-12-15:
@LaWfulN there are some instructions I found here that talk you through how to manually update and install the module with the details @avidesk provided:
[](https://github.com/jgarff/rpi_ws281x/issues/483)

presinnovation, 2023-12-18:
I got it running on my Pi 3A+ v1.1 and have opened PR #531 with this change.

daprezjer, 2024-06-27:
> @LaWfulN there are some instructions I found here that talk you through how to manually update and install the module with the details @avidesk provided:

@rnlgreen it looks like your link didn't post correctly.

rnlgreen, 2024-06-28:
@daprezjer sorry about that, I think it was just this other issue:
[https://github.com/jgarff/rpi_ws281x/issues/483](url)