jgarff/rpi_ws281x #528: Support for Raspberry Pi 5

Puetz, 2023-11-04:
Hi,

I just got a Raspberry Pi 5 and am getting an error that my hardware revision is not supported.

Can someone please add support for the Raspberry Pi 5 or give me a guideline on how to do it myself?
I'd greatly appreciate it.

Best regards,
Thomas

Gadgetoid, 2023-11-04:
The Pi 5 has significant hardware changes that might make some of this library difficult or impossible to migrate. I’m not qualified to know exactly how bad it is, but the changes are significant enough that a straight hardware ID addition isn’t going to cut it. We need one of the wizards involved in writing or extending the core functionality of rp_ws281x to step in and give us the lowdown.

I’ve talked to folks at Pi about creating a PIO-backed WS281X library for RP1, but there’s no telling when it if that - or the tools to make it happen - will materialise.

jgarff, 2023-11-04:
I've definitely been wanting to add support for that, if possible.
Unfortunately, I don't yet have a RPi 5 to actually work on.

Jeremy

Gadgetoid, 2023-11-06:
> I've definitely been wanting to add support for that, if possible.

Wow, of all the things I expected- it wasn't the big guns! I hope you're well.

> Unfortunately, I don't yet have a RPi 5 to actually work on.

That's something I can probably fix if you want to give support a shot!

ubopod, 2023-11-08:
@jgarff If you are in Us/Canada, I can send you a RPi5 to support the port and any other way I can support this. I am getting a similar issue:

```
raise RuntimeError(
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
```

ebraraktas, 2023-11-11:
Hi @jgarff , I tried to add HW spec and debug the library (with `main.c` after some modification for my led strip). I added the spec below:
```c
    //
    // Raspberry Pi 5
    //
    {
        .hwver = 0xc04170,
        .type = RPI_HWVER_TYPE_PI5,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 5 Model B - 4GB 1.0"
    },
```
where `RPI_HWVER_TYPE_PI5` is defined as `4`. And it failed to allocate memory in the lines below:
https://github.com/jgarff/rpi_ws281x/blob/1f47b59ed603223d1376d36c788c89af67ae2fdc/ws2811.c#L939-L940
here is the `dmesg` output for that request:
```
[ 1884.471030] raspberrypi-firmware soc:firmware: Request 0x0003000c returned status 0x80000001
```
And ioctl errno corresponds to `Invalid argument`. 

I know nothing about RPi (this is my first device) and mailboxes etc. but I tried to run a simpler code (`info.c`) in [this library](https://github.com/Terminus-IMRC/mailbox) to identify the issue, and here is the output (I added `Cannot get ...` logs after failure):
```
root@raspberrypi:/home/ebraraktas/Projects/mailbox# ./build/test/info 
Firmware revision: Oct 18 2023 17:30:17
/home/ebraraktas/Projects/mailbox/src/fd.c:52:mailbox_property: error: ioctl: IOCTL_MBOX_PROPERTY: Invalid argument
Cannot get board model
/home/ebraraktas/Projects/mailbox/src/fd.c:52:mailbox_property: error: ioctl: IOCTL_MBOX_PROPERTY: Invalid argument
Cannot get board revision
/home/ebraraktas/Projects/mailbox/src/fd.c:52:mailbox_property: error: ioctl: IOCTL_MBOX_PROPERTY: Invalid argument
Cannot get mac address
Board serial: 0xc77d11596528bd61
ARM memory: 0x3f800000 bytes at 0x00000000
VC memory:  0x02100000 bytes at 0xfdb00000
```
And `memflag.c`([link](https://github.com/Terminus-IMRC/mailbox/blob/master/test/memflag.c)) failed to allocate memory with the same `errno`, too.

I don't know if any of these helps to diagnose the issue, but I wanted to post my findings. If this issue seems to be resolved quickly, I would like to help it (but it seems way beyond my knowledge).

jgarff, 2023-11-12:
That would be fantastic!  I'm hoping to get a little time over the holidays
to work on this a bit.

ubopod, 2023-11-12:
Do you mind sending me an email to mehrdad@getubo.com so that I can arrange to have one shipped to you?

jgarff, 2023-11-27:
I've now got hardware to play with.  Many many thanks to @ubopod for the very generous donation!  I've started looking into it, and have found the following so far:

- Changing the hardware device ID is unfortunately insufficient to add support
- Using mailbox messages to allocate DMA coherent memory that we can access from userspace doesn't work apparently.  I'm investigating, but this is likely due to API changes when communicating with the Videocore VII firmware.
- The GPIOs now runs through the new RPi5 RP1 chipset.  Because of this they are not compatible with the previous PWM IP in the Broadcom SoC.  Although, the RP1 has some neat IP that we can use instead.  I'm looking into using the PIO controller in the RP1 along with DMA.  I just need to get a cache free userspace buffer to work from.

I'll keep you guys posted with updates.  Please be patient though, as I'm fairly busy, but hoping  to work more on this over the holidays.

jgarff, 2023-12-02:
Just a quick update.
- I figured out how to get uncached memory from userspace _without_ using the videocore mechanism.  This should actually work for all versions of the raspberry pi with newer kernels, and it's much cleaner, so I'm going to rip out the mailbox interface and replace it globally.  Once this is done I'll start on the RP1 DMA/PIO.

mbevand, 2023-12-05:
I was playing for a few hours with attempting a port to work on the Pi 5.

I can confirm that most of the videocore firmware mailbox interfaces have been ripped out (https://forums.raspberrypi.com/viewtopic.php?t=359477) so definitely you need to drop that whole mechanism.

Another thing is the peripheral base address is now 0x107c000000 (instead of 0xfe000000 as on the Pi 4). I believe the new RP1 DMA/PIO will have to be configured through registers in that space, and because it's a base address over the 32-bit limit, you may need to `#define _FILE_OFFSET_BITS 64` so that you can access it through `/dev/mem`

jgarff, 2023-12-05:
Thanks @mbevand for the info.  Since the RP1 is connected via PCIe, doesn't the base of the registers land in a PCIe BAR?

	Region 0: Memory at 1f00410000 (32-bit, non-prefetchable) [size=16K]
	Region 1: Memory at 1f00000000 (32-bit, non-prefetchable) [virtual] [size=4M]
	Region 2: Memory at 1f00400000 (32-bit, non-prefetchable) [size=64K]

Are they remapping that somewhere else (0x107c000000?) in the rp1 driver?

jgarff, 2023-12-05:
I'm not worried about a conflict.  What I'm suggesting is that the peripherals that are built into the bcm2712 (dma/pcm/pwm/etc.) are no longer being used on the Pi5.

Instead the GPIOs are now wired to the RP1, not the SoC.  The RP1 in turn is available to the SoC over PCIe, with all the registers mapped inside a RP1 BAR.  Basically we can't use the peripherals in the 2712 anymore as they aren't wired to the actual pins anymore.

mbevand, 2023-12-05:
Doh! I wasn't thinking straight. You are right, of course. The register region at `0x107c000000` is useless to us. As you say, the RP1 PWM registers should be mapped inside a BAR region...

jgarff, 2023-12-05:
No worries.  I wish you were right.  This would be much easier if the pins were still connected to the SoC IP.

mbevand, 2023-12-06:
I don't know if that's useful or if you already know it, but the BAR can be accessed through sysfs. I wrote this sample code that shows how to read the FIFO_CTRL register of PWM0 on RP1. On my Pi 5 it prints `FIFO_CTRL: 20000` since that's the default value of this register, as per page 39 of https://datasheets.raspberrypi.com/rp1/rp1-peripherals.pdf :

 ```
#include <sys/ioctl.h>
#include <sys/mman.h>
#include <assert.h>
#include <fcntl.h>
#include <unistd.h>
#include <stdlib.h>
#include <stdio.h>
#include <stdint.h>

#define OFFSET_PWM0 0x98000
#define OFFSET_PWM1 0x9c000

#define PWM_GLOBAL_CTRL 0x00
#define PWM_FIFO_CTRL   0x04

uint32_t my_read(void *base, ssize_t count, uint32_t offset)
{   
    assert(count == 4);
    return *(uint32_t *)((uint8_t *)base + offset);
}

int main() {
    const char *fname = "/sys/bus/pci/devices/0000:01:00.0/resource1";
    void *base;
    int fd;
    uint32_t reg;

    if (-1 == (fd = open(fname, O_RDWR)))
        perror("open"), exit(1);
    base = mmap(NULL, 0x400000, PROT_READ | PROT_WRITE, MAP_SHARED, fd, 0);
    if (base == MAP_FAILED)
        perror("mmap"), exit(1);
    if (-1 == close(fd))
        perror("close"), exit(1);
    reg = my_read(base, 4, OFFSET_PWM0 + PWM_FIFO_CTRL);
    printf("FIFO_CTRL: %x\n", reg);
    return 0;
}
```

jgarff, 2023-12-06:
Thanks.  I'm well aware of these mechanisms.  The tricky part isn't
register access.  It's the dma.

O-O-O-84, 2023-12-31:
Hey guys, i've got the same issue. is there already a solution? Maybe an update for the RPI_WS281x?

ikke01, 2024-01-02:
> I've now got hardware to play with. Many many thanks to @ubopod for the very generous donation! I've started looking into it, and have found the following so far:
> 
> * Changing the hardware device ID is unfortunately insufficient to add support.
> * Using mailbox messages to allocate DMA coherent memory that we can access from userspace doesn't work apparently.  I'm investigating, but this is likely due to API changes when communicating with the Videocore VII firmware (new in the RPi5).  I could write a small kernel driver to map this type of memory to userspace, but I'm hoping to not have to do that.
> * The GPIOs now run through the new RPi5 RP1 chipset.  Because of this they are not compatible with the previous PWM IP in the Broadcom SoC.  Although, the RP1 has a neat IP that we can use instead.  I'm looking into using the PIO controller in the RP1 along with DMA.  I just need to get a cache free userspace buffer to work from.
> 
> I'll keep you guys posted with updates. Please be patient though, as I'm fairly busy, but hoping to work more on this over the holidays.

Hallo Jeremy, did you make any progress on this for the PI5, we are looking forward to it if you can fix it to get it working again.  Thanks Raymond

jgarff, 2024-01-02:
Hi Raymond, I was able to work on this some over the holidays.  I have some open questions in a developer forum regarding the DMA that I'm hoping can be answered.  Unfortunately, the specs available thus far are lacking in specific functionality of the RP1.  Some of which are tied up behind non-public specifications for intellectual property from third parties used in the RP1.

ikke01, 2024-01-02:
Thank you for your fast response, let hope they respond, this makes the use of the pins not very usefull anymore if the cannot be addressed as in the PI4 and older versions.

Raymond

mehrdadfeller, 2024-01-02:
@jgarff Thanks again for the efforts! I hope the necessary information come out soon. If you link to relevant developer forum questions here, we can collectively try to get that information for Raspberry Pi. I have a few contacts in the commercial side and can also ask around to get more attention to it.

jgarff, 2024-01-03:
That would be great @mehrdadfeller.  The specific questions I have are as follows:
- Need to get access to the synopsys DMA controller documentation.
- Need to get the channel mappings for each peripheral (PWM/PIO/etc.) as it relates to the DMA.  These are typically used for things like flow control/backpressure/etc.  This is how the DMA controller knows that the given channel FIFO is full and whatnot.  It's hard to know the specifics without the DMA docs though.
- There is a linux synopsys DMA driver in the kernel.  I've gone through some of the code.  I may end up having to write a kernel driver for this though if I want to use the same driver, but still need the mapping info.

jgarff, 2024-01-03:
https://forums.raspberrypi.com/viewtopic.php?t=361457

GuyPerets106, 2024-01-09:
Hi , I'm following this issue. First of all , big thanks to @jgarff for working on this issue.
Secondly , I saw this product from Adafruit -  https://learn.adafruit.com/adafruit-neodriver-i2c-to-neopixel-driver?view=all that seemingly should allow NeoPixel to run on devices and microcontrollers which does not support it. Maybe I don't understand all the technical details , but someone can verify it can work on the RPi 5 until the current issue resolved? 
Thanks.

madmongoose, 2024-01-09:
> Hi , I'm following this issue. First of all , big thanks to @jgarff for working on this issue. Secondly , I saw this product from Adafruit - https://learn.adafruit.com/adafruit-neodriver-i2c-to-neopixel-driver?view=all that seemingly should allow NeoPixel to run on devices and microcontrollers which does not support it. Maybe I don't understand all the technical details , but someone can verify it can work on the RPi 5 until the current issue gets resolved? Thanks.

I've checked all buses, unfortunately it doesn't work

sudo i2cdetect -l
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-4	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-11	i2c       	107d508200.i2c                  	I2C adapter
i2c-12	i2c       	107d508280.i2c                  	I2C adapter
madmongoose@mmg-rpi:~ $ vim test-led2.py 
madmongoose@mmg-rpi:~ $ python test-led2.py 
/home/madmongoose/.local/lib/python3.11/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py:30: RuntimeWarning: I2C frequency is not settable in python, ignoring!
  warnings.warn(
Traceback (most recent call last):
  File "/home/madmongoose/.local/lib/python3.11/site-packages/adafruit_bus_device/i2c_device.py", line 175, in __probe_for_device
    self.i2c.writeto(self.device_address, b"")
  File "/home/madmongoose/.local/lib/python3.11/site-packages/busio.py", line 207, in writeto
    return self._i2c.writeto(address, buffer, stop=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/madmongoose/.local/lib/python3.11/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 60, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/home/madmongoose/.local/lib/python3.11/site-packages/Adafruit_PureIO/smbus.py", line 303, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/madmongoose/.local/lib/python3.11/site-packages/adafruit_bus_device/i2c_device.py", line 181, in __probe_for_device
    self.i2c.readfrom_into(self.device_address, result)
  File "/home/madmongoose/.local/lib/python3.11/site-packages/busio.py", line 197, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/madmongoose/.local/lib/python3.11/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 67, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end - start)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/madmongoose/.local/lib/python3.11/site-packages/Adafruit_PureIO/smbus.py", line 170, in read_bytes
    return self._device.read(number)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/madmongoose/test-led2.py", line 7, in <module>
    ss = seesaw.Seesaw(i2c, addr=0x60)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/madmongoose/.local/lib/python3.11/site-packages/adafruit_seesaw/seesaw.py", line 147, in __init__
    self.i2c_device = I2CDevice(i2c_bus, addr)
                      ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/madmongoose/.local/lib/python3.11/site-packages/adafruit_bus_device/i2c_device.py", line 62, in __init__
    self.__probe_for_device()
  File "/home/madmongoose/.local/lib/python3.11/site-packages/adafruit_bus_device/i2c_device.py", line 184, in __probe_for_device
    raise ValueError("No I2C device at address: 0x%x" % self.device_address)
ValueError: No I2C device at address: 0x60

madmongoose, 2024-01-09:
I checked as sudo and I have another error. I also tried to install colorwheel or rainbowio. But unfortunately...

madmongoose@mmg-rpi:~ $ sudo python test-led.py
Traceback (most recent call last):
  File "/home/madmongoose/test-led.py", line 8, in <module>
    from rainbowio import colorwheel
ModuleNotFoundError: No module named 'rainbowio'

danic85, 2024-01-09:
> Hi , I'm following this issue. First of all , big thanks to @jgarff for working on this issue. Secondly , I saw this product from Adafruit - [...]

I'm using this on a Raspberry Pi 5 currently, so I can confirm it does work.
https://www.makerforge.tech/posts/pi5-build-log-4/#neopixel-i2c-support

madmongoose, 2024-01-09:
> I'm using this on a Raspberry Pi 5 currently, so I can confirm it does work. https://www.makerforge.tech/posts/pi5-build-log-4/#neopixel-i2c-support

Hi! My bad, I installed not as sudo. Could you please provide your script for test?

danic85, 2024-01-09:
> Hi! My bad, I installed not as sudo. Could you please provide your script for test?

The sample code in Guy's link works for me - I have a more complex implementation but that isn't needed for the basic functionality.

Would still be great to see this repo working on the Pi 5 as the i2c module is a bit of a workaround for the issue for me.

madmongoose, 2024-01-09:
> The sample code in Guy's link works for me - I have a more complex implementation but that isn't needed for the basic functionality.
>
> Would still be great to see this repo working on the Pi 5 as the [...]

Thanks for reply.

Maybe it doesn't work for me because I connect led through this board to rpi5.

[image]

What bus number you used?

Gadgetoid, 2024-01-09:
> the specs available thus far are lacking in specific functionality of the RP1

It's not much use in the near term, but I am discussing (with Pi) the feasibility of WS281x support, or perhaps some userspace tools and interfaces to make it possible, that leverages the RP1's PIO. It's not a "no", but more a "this will be complicated and requires some thought and a fair bit of work."

I remain optimistic, since there's a desire to show off the RP1's PIO, that the Pi 5 will be able to support WS281x strips on *arbitrary* pins using a simple PIO program and the same DMA-a-bunch-of-carefully-crafted-bytes trick we use currently. This could potentially even extend to custom PIO code to emit the correct pulses for 1s/0s, but right now I can only really speculate based on a few brief discussions of our requirements.

Sorry I didn't get a Pi 5 out to you quickly- I am absolutely mired in getting everything else to work!

mehrdadfeller, 2024-01-09:
@Gadgetoid That's great! I am trying to get some support from Raspberry Pi on this as well and I am also optimistic that they will come out with something cool. 

BTW, do you know what clock speed these PIOs can toggle at? If RP1 on Pi 5, could support DMA on arbitrary PINs, does that mean you can drive multiple long strips in parallel as well? I wonder what is the bottleneck...

@danic85 @madmongoose: I might be mistaken but I think I2C speed (400KHz) could limit how fast you can drive LED strips but it works for many applications. This won't be noticeable if you are showing static patterns but if you are transitioning fast between colors/patterns or you are piping too much on long strips, DMA can offer must faster transfer speeds. Also, with DMA you don't need the extra hardware (MCU) in between.

Thee's another type of LED that Adafruit calls DotStart that you do not have to do DMA. Adafruit NeoPixel is the terminology used for WS281X LEDs. There's a good comparison [here](https://learn.adafruit.com/adafruit-dotstar-leds/overview#dotstars-vs-neopixels-2002908).

I have always used WS281X LEDs because they are much cheaper (<$0.01 at large volumes), you can buy them easily in various types of packagings (right angle, fused lens, etc).

madmongoose, 2024-01-09:
@mehrdadfeller hi! I can't use a led without MCU in between, because https://github.com/Freenove/Freenove_Robot_Dog_Kit_for_Raspberry_Pi

I built this build instead of the recommended rpi4 on rpi5, corrected the code a bit, replaced GPIO with gpiozero. Servo drives, buzzer, ultrasonic work well. All that's left is to fix the led.
=)

orsemdev, 2024-01-15:
> > > the specs available thus far are lacking in specific functionality of the RP1
> > 
> > 
> > It's not much use in the near term, but I am discussing (with Pi) the feasibility of WS281x support, or perhaps some userspace tools and interfaces to make it possible, that leverages the RP1's PIO. It's not a "no", but more a "this will be complicated and requires some thought and a fair bit of work."
> > I remain optimistic, since there's a desire to show off the RP1's PIO, that the Pi 5 will be able to support WS281x strips on _arbitrary_ pins using a simple PIO program and the same DMA-a-bunch-of-carefully-crafted-bytes trick we use currently. This could potentially even extend to custom PIO code to emit the correct pulses for 1s/0s, but right now I can only really speculate based on a few brief discussions of our requirements.
> > Sorry I didn't get a Pi 5 out to you quickly- I am absolutely mired in getting everything else to work!
> 
> Thanks Gadgetoid! I was planning on actually using the new PIO to do this as it will be more flexible, allow for more GPIO ports, and sounds fun anyway. We can also leverage some of the microcode for it from the microcontroller projects.
> 
> The questions I have open (without reply unfortunately) are with regard to getting the DMA working with the PIO controller. Would you mind involving me in your discussions with Raspberry Pi? I would greatly appreciate it, and would very much like to be involved. Thanks!

Hi there,

I have just switched from my Raspberry Pi 2 --> Raspberry Pi 5 and my ws2812b scripts are no longer running without an error (Hardware revisions not supported). I assume that the problem is the outdated library, do you have any updates on the update for Raspberry Pi 5?

Thanks.

ubopod, 2024-01-15:
@orsemdev this issue is blocked due to lack of documentation on certain features of RP1 chip that handles IO on Raspberry Pi 5. We need to wait til the specs are released.

felixfeliciz, 2024-01-17:
Until then, is there any way to use the WS2812B with a limited range of functions, such as simple switching on and off or simple color changes without fancy gradients or animations?

felixfeliciz, 2024-01-17:
> Until then, is there any way to use the WS2812B with a limited range of functions, such as simple switching on and off or simple color changes without fancy gradients or animations?

without using Adafruit NeoDriver - I2C to NeoPixel Driver

jgarff, 2024-01-17:
Unfortunately no.  Setting any color at all requires transferring a
complicated sequence of signals over the wire.  The timing required for
these signals isn't possible from a software only implementation on Linux
due to context switches, driver interrupts, etc..

ubopod, 2024-01-17:
I have not personally tried that but you could potentially use SPI option
which should not pose any issues on Pi 5. I believe the issue is using DMA
mainly which can be used to do fast PWM/PCM.

Since I use an LCD display on SPI bus already that is not an option for me
but that could work in your case if you are not currently using it to drive
something else.

jgarff, 2024-01-17:
That's a great idea as well!  I haven't tried it, and I know the SPI
hardware is also different on the RPi5, but in theory it should be able to
output the right waveform.

cleverca22, 2024-01-29:
> Need to get access to the synopsys DMA controller documentation.

https://github.com/cleverca22/rp1-kernel-example/blob/23fcd99ef40d2968d22f29cd808634814cc6d584/rp1-kernel-test.c#L80-L106

this is an example of using all of the proper kernel api's to manage the dma controller

> Need to get the channel mappings for each peripheral (PWM/PIO/etc.) as it relates to the DMA. These are typically used for things like flow control/backpressure/etc. This is how the DMA controller [...]

https://github.com/librerpi/rp1-lk/blob/757fbcb8f2554643989aac719e0cdb56a052f247/platform/rp1/pio.c#L48
this is a pio uart example that pairs up with that 

https://github.com/MichaelBell/rp1-hacking/blob/main/PIO.md
this documents most of the pio differences

https://github.com/MichaelBell/rp1-hacking/blob/main/launch_core1/core1_test.c
and this temporarily hijacks core0 to get rp1-lk running on core1

the biggest issue, is that the PIO control registers are up at `0xf000_0000` in the private peripherals of the cortex-m3 pair
and there is currently no proper way to gain control of either M3
`core1_test.c` will overwrite some of the assembly in core0's firmware, so when a main control loop gets to that function, it runs some custom assembly, which can kick-start core1

rp1-lk is then a fairly standard cortex-m style binary, with the initial sp/cp at the start, and `core1_test.c` will launch that on core1
that then configures the PIO for whatever your desired task is, and enables DREQ

then the rp1-kernel-example can rig up dma and perform copies to/from the fifo's

i'm currently working on a logic analyzer type project, and i can get it to capture up to 581mbit from PIO

you can also find me over at ##raspberrypi-internals on libera if you want some more realtime questions

jgarff, 2024-01-29:
This is great information @cleverca22, thank you.  It's unfortunate however that the Cortex-M3's are required to read/write the PIO registers, which necessitates the need to get them involved.  I'll dig into the links you've provided.  Thanks again.

ethomas997, 2024-02-02:
I hope support for the Pi 5 can be resolved.  Over at our [RotorHazard Race Timer](https://github.com/RotorHazard/RotorHazard) project we depend on this library for running LED strips and panels.

cleverca22, 2024-02-05:
@jgarff having a second look at the rp1 datasheet, the PWM block has a serializer mode, like the bcm SoC's

so it should just be a matter of rigging up dma to feed duty-cycles to the pwm controller and putting it in serializer mode
the pwm dreq channel is in `include/dt-bindings/mfd/rp1.h` along with all of the other dreq's

jgarff, 2024-02-05:
Yeah, I've thought about that as well.  Do you think it's possible to use the DMA to write to the PIO registers to setup the PIO without having to go through the Cortex-Ms?  I don't like messing with the Cortex-M's if they're used for things like fan speed control, resets, and whatnot.  Obviously using the DMA to write registers is inefficient, but would mean not having to use the Cortex-Ms.

MichaelBell, 2024-02-05:
Unfortunately as far as we can tell the PIO can only be configured directly from the Cortex-Ms.

Given what @cleverca22 and I have done so far it would be fairly easy to make a kernel module that patched the code on the Cortex-M to configure the PIO with a ws2812 pio program, and then exposed a character device that you could use to DMA data to the PIO SM.

But the patching is very fragile and likely break with future firmware updates - it's not really suitable for something that lots of people are likely to use.  We really need Raspberry Pi to give us some kind of mechanism to safely perform this configuration - until then it's possible to make a demo, but not really something that would be wise to distribute widely.

In the meantime, PWM ought to be a reasonable stop gap solution, although it obviously only works on some of the available header pins.

cleverca22, 2024-02-06:
> I don't like messing with the Cortex-M's if they're used for things like fan speed control, resets, and whatnot.

luckily, only core 0 is used for the fan rpm (and some unknown pcie duties)
the fan pwm is entirely linux hitting pwm regs directly

core 1 is entirely unused
but you need to temporarily hijack core0, to get core1 bootstrapped

cleverca22, 2024-02-06:
oh, and i forgot about @G33KatWork 's older stuff, before @MichaelBell found the core0 hack

https://github.com/G33KatWork/RP1-Reverse-Engineering

basically, you can just hard-reset the entire RP1, and re-load the core0 and core1 firmware over i2c, and reboot it
that would have better compatibility with the core0 firmware
both because your not patching it, and because you choose which core0 version is in use

but linux currently doesnt handle removal of the rp1 cleanly
so you need a dtoverlay that disables the rp1, reset it, then a runtime dtoverlay to allow rp1 access
and you can only do that once per host boot, so linux has to reboot to do it again

mehrdadfeller, 2024-02-06:
Would it be possible to use cortex m3 to run a custom firmware at boot
time? If so, this can be quite valuable and can solve several problems:

For example the MCU could drive GPIO and SPI/I2C buses before the Pi boots
up (since the boot up can take sometimes around 20 seconds). For example if
you have screen, mcu can show a splash or some message to the user.

More importantly, MCU could be used as a fallback options to perform some
sort of system recovery if OS is corrupt...

These are ofcourse not related to the specific use case here though

cleverca22, 2024-02-06:
the pi5 goes thru a few phases during boot
first it loads a `bootcode.bin` from SPI flash, and executes it, that brings the dram controller online
then it loads `bootmain.elf` from SPI, and executes it, that deals with booting the whole system, and replaces the old `start(4).elf`
bootmain then has an rp1 firmware baked within its `.rodata`, it will push that over i2c to the rp1, and confirm its working, before moving on to starting the arm core

that whole chain, is signed with an unknown mechanism at the root, so any attempt to modify it will just result in the pi5 not booting

but once something is running on the main cortex-a76 core, it can just take over the gpio, hard-reset the RP1, and push anything it wants over i2c

@G33KatWork 's repo includes examples of doing that from linux, and i believe its possible to push the original core0 firmware, plus something custom
in theory, you could modify u-boot into doing that as well, so core1 is doing something before linux has even been read from the card

but it cant really be used as a fallback, because you need an arm kernel on the bootfs, to hijack the RP1

but with the pi0-pi3 family, you can run https://github.com/librerpi/lk-overlay as a firmware replacement
then you can add whatever custom splash screen you want on any interface you can write drivers for

sadly, the pi5 has the VPU locked down even more then ever, and the open firmware may just never be an option for the 5

jgarff, 2024-02-07:
It's interesting that they opted for loading the RP1 over I2c instead of having a small ROM.  Makes sense though in that they can always program it from the bootloader of the BCM.

cleverca22, 2024-02-07:
having decompiled the RP1 boot rom, it supports booting in either i2c-slave mode, or spi-master mode
so they could have put a second spi chip on the board, and bootstrap the rp1 from that

but that would then drive up the cost of the whole assembly
same reason they ditched the vl805 spi chip

willmorris04, 2024-02-12:
> It's interesting that they opted for loading the RP1 over I2c instead of having a small ROM. Makes sense though in that they can always program it from the bootloader of the BCM.

Hi jgarff, I appreciate the effort you are making to update the ws281x library. I was wondering if there is a simple way to control the led I have. I only want to turn it on and off at full brightness white. I'm a beginner to microprocessors and would really appreciate some advice. Thanks

jgarff, 2024-02-12:
Unfortunately, it's not easy to simply turn on one LED, or any color without sending the wire protocol.  The same basic logic is needed to turn on 1 or 1000.

stephinson, 2024-02-17:
> Hi jgarff, I appreciate the effort you are making to update the ws281x library. I was wondering if there is a simple way to control the led I have. I only want to turn it on and off at full [...]

You can try SPI to turn on/off LEDs

https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel_SPI/issues/37#issuecomment-1871698900

Lukas200011, 2024-02-22:
I would also be very interested to get ws28xx to work with a Raspberry Pi 5. What is the current situation of fixing this issue  and future plans? I don't want to rush you with this post, I am just curious. :) 
Have a nice week!

jgarff, 2024-03-12:
Sadly, the PIO IP in the RP1 doesn't allow for direct access from Linux, and is instead only controllable through the RP1 Cortex-M's.  While that is the best long term goal, it's far complicated, especially in a easily deployable library.  In the short term I've written a kernel driver using PWM instead.  It's not quite done, but I'm getting closer.  I got some waveforms on my scope last night.

Unfortunately, the PWM won't be able to control 10 separate strips at the same time.  The RP1 has 2 PWM instances, but one is dedicated to fan control, leaving us with only one 4 channel PWM.  In this case 4 strips would be the max.

Gadgetoid, 2024-03-12:
> Sadly, the PIO IP in the RP1 doesn't allow for direct access from Linux, and is instead only controllable through the RP1 Cortex-M's.

AFAIK first party support for WS281X is still a distinct possibility but we can only sit on our hands and wait, and have no idea exactly how it might perform.

hannescam, 2024-03-13:
> Sadly, the PIO IP in the RP1 doesn't allow for direct access from Linux, and is instead only controllable through the RP1 Cortex-M's. While that is the best long term goal, it's far more complicated, especially in an easily deployable library. In the short term I've written a kernel driver using PWM instead. It's not quite done, but I'm getting closer. I got some waveforms on my scope last night.
> 
> Unfortunately, the PWM won't be able to control 10 separate strips at the same time. The RP1 has 2 PWM instances, but one is dedicated to fan control, leaving us with only one 4 channel PWM. In this case 4 strips would be the max.

Okay I might switch to USB-SPI bridge chips connected via USB through an USB HUB to the RPi5 with your driver (I saw that your driver supports SPI) but when the driver is ready (full PIO support) i will switch to the direct output of the RPi5

Gadgetoid, 2024-03-13:
Would be interesting to know if a plain SPI bridge chip would have the throughput required. I’d just use a bunch of RP2040 (Pico) boards since they could update potentially thousands of LEDs continuously from an internal buffer.

Had no problem running ~12,000 LEDs split across 20 USB devices from one Pi using RP2040s- https://youtu.be/cm83RIhDbwo?si=fFB7bMm1-oSSfV8Y&t=10m48s

[104 more replies]