jgarff/rpi_ws281x #529: Raspberry Pi 5B with Neopixel: Hardware revision is not supported

josh-msn, 2023-11-25:
Hey,

I am relatively new to coding and want to control a Neopixel stick with the Raspberry Pi 5. I get this error: 

raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
Segmentation fault

I tried this solution: 
https://github.com/rpi-ws281x/rpi-ws281x-python/issues/86

But I am not sure on how to get the hardware version and install the edited package. I hope someone can help me.

Any help / tips will be greatly received :-)

Joshua

tomaszszymanski129, 2023-11-27:
This is a duplicate of https://github.com/jgarff/rpi_ws281x/issues/528
Unfortunately the Pi5 is not supported yet, hardware is very different from older Pi boards and the lib has to be adjusted, there might be some plans to do that (see the issue), I'm counting on that as well.

jgarff, 2024-10-16:
Please see the following:
https://github.com/jgarff/rpi_ws281x/wiki/Raspberry-Pi-5-Support

Hope that helps.

closed 2024-10-16 by jgarff