jgarff/rpi_ws281x #535: Before I re-invent the wheel...

davthomaspilot, 2024-04-04:
I want to avoid running my big, Qt app as root but I need to use the ws2811 library.  

I've been using it for years in several projects and it works great--except now I can play with permissions (Bookworm) to avoid needing root access.

So, I thought I write some code that runs at root and accepts socket commands for ws2811_init, ws2811_fini, ws2811_render, etc.

Has anyone already done this?

hannescam, 2024-04-15:
I have written an SPI only NeoPixel driver and it doesn't need root.
But I don't know about this library.
Just as a warning: If you want to use my library you can but it is pretty bare-bones but I will accept reasonable feature requests

mkrizmancic, 2025-07-11:
Hi, I had a very similar problem, so I decided to write a small [library](https://github.com/mkrizmancic/rpi_ws281x_pylib) consisting of a root server that sets the LEDs and a non-root client that sends commands, only at the time I decided to use [pyzmq](https://github.com/zeromq/pyzmq) for communication.

It's probably not the best, but if somebody comes across this issue, you may find it useful.