jgarff/rpi_ws281x #544: `version` file causes errors when a file in the project includes <version>

CalcProgrammer1, 2024-08-06:
https://en.cppreference.com/w/cpp/header/version

<version> is a standard header name in C++20, so there is the possibility that files within a given project include it.  The `version` file in this project can cause issues if this happens as it is not valid C++ code and instead just a text file that contains a version number.  I would recommend changing this file's name so that it doesn't clash with a known C++ standard header name.

When pulling this library into my project, I am running into this issue because the nlohmann json library I also use has #include <version>.  Here is a build log for reference:

```
In file included from /builds/CalcProgrammer1/OpenRGB/dependencies/json/json.hpp:2404,
                 from /builds/CalcProgrammer1/OpenRGB/SettingsManager.h:16,
                 from /builds/CalcProgrammer1/OpenRGB/qt/DetectorTableModel.cpp:11:
/builds/CalcProgrammer1/OpenRGB/dependencies/rpi_ws281x/version:1:1: error: too many decimal points in number
    1 | 1.1.0
      | ^~~~~
```

I've included the rpi_ws281x directory in my project's include directories list, which is necessary for the actual header files to be picked up.  The alternative would be to move all the include files into an include/inc directory so that `version` is not picked up when the path is included.

penfold42, 2024-10-27:
Or rename version to .version as per #527 ?

jgarff, 2024-10-27:
I'm not a fan of using dotfiles as they don't normally show up in a directory listing.  Shouldn't you be including this file as extern "C" anyway?