jgarff/rpi_ws281x #545: ws2811_init failed Pi5 support

zytegalaxy, 2024-10-13:
I followed the steps in the document below and everything went fine until the last step.

https://github.com/jgarff/rpi_ws281x/wiki/Raspberry-Pi-5-Support

In the last step when I try to run the test I get the `Hardware revision is not supported`:

```sh
ubo@ubo-3w:~/rpi_ws281x $ sudo ./test 
sudo: unable to resolve host ubo-3w: Name or service not known
ws2811_init failed: Hardware revision is not supported
```

I suspect there might be minute differences between various Pi5 devices that make this fail? 

I tried forcing it to specific values (bypassing the `ws2811->rpi_hw = rpi_hw_detect();`) but I couldn't get the LEDs to light up.

```c
    // Force Pi 5 configuration
    static const rpi_hw_t pi5_hw = {
        .type = RPI_HWVER_TYPE_PI5,
        .hwver = 0xa05000,
        .periph_base = 0x7c000000,
        .videocore_base = 0xfc000000,
        .desc = "Pi 5"
    };

    ws2811->rpi_hw = &pi5_hw;
    rpi_hw = ws2811->rpi_hw;
``` 

Please let me know if I can provide detailed platform specific information that can help find the differences.

```shell
cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x4
CPU part        : 0xd0b
CPU revision    : 1

Revision        : c04170
Serial          : acb62badbaabaf02
Model           : Raspberry Pi 5 Model B Rev 1.0
```

jgarff, 2024-10-13:
I just added your hardware to the config.  Also please note that if you add
another revision, you need to make sure to set the periph_base and
videocore_base to zero.

Lastly, please double check if you're using pin 18.  I have a suspicion
that you're using pin 12 instead.  If that's the case, you will need to
adjust the calls to rp1_ws281x_pwm_init(0,0) in the rp1_ws281x_pwm.c and
recompile.  Pin 12 is associated with channel 0 in the PWM hardware.   I
will address that when I add multi channel support soon.

Then you need to use the "sudo pinctrl set 12 a0 pn" command instead of the
one listed in the wiki.

Hope that helps,
Jeremy

zytegalaxy, 2024-10-13:
@jgarff thanks so much for addressing the issue. Now I know how you are setting the config. If I run into a similar issue with a different version of Pi5, I will do a pull request to it to the repo. I confirm your latest change addresses the hardware version detection issue.

I am still going through the code to better understand the details. I made the suggested changes and I was able to get the LEDs light up but I am not seeing the correct pattern yet. That's probably an issue with my configuration and/or how I am setting things up. 

I am going to close this issue but might open a separate one if I can't come to a conclusion by myself on getting the patterns light up correctly.

closed 2024-10-13 by zytegalaxy